ni/nidaqmx-python #417: PCIe-6321 hardware failure when using hardware timing

williamsnider, 2023-06-19:
When I run the following code (snippet from the ao_voltage_hw_timed.py example), I get an error:
```
"""Example of AO voltage hw operation."""
import nidaqmx

with nidaqmx.Task() as task:
    task.ao_channels.add_ao_voltage_chan("Dev1/ao0")

    task.timing.cfg_samp_clk_timing(1000)

    print("1 Channel N Samples Write: ")
    print(task.write([1.1, 2.2, 3.3, 4.4, 5.5], auto_start=True))
    task.wait_until_done()
    task.stop()
```

error message:
```
1 Channel N Samples Write: 
Traceback (most recent call last):
  File "/home/oconnorlabmatlab/Code/libfranka/python/nidaq/ao_voltage_hw_timed.py", line 11, in <module>
    print(task.write([1.1, 2.2, 3.3, 4.4, 5.5], auto_start=True))
  File "/home/oconnorlabmatlab/miniconda3/envs/robot_venv/lib/python3.9/site-packages/nidaqmx/task.py", line 1274, in write
    return _write_analog_f_64(
  File "/home/oconnorlabmatlab/miniconda3/envs/robot_venv/lib/python3.9/site-packages/nidaqmx/_task_modules/write_functions.py", line 28, in _write_analog_f_64
    check_for_error(error_code, samps_per_chan_written=samps_per_chan_written.value)
  File "/home/oconnorlabmatlab/miniconda3/envs/robot_venv/lib/python3.9/site-packages/nidaqmx/errors.py", line 181, in check_for_error
    raise DaqWriteError(error_buffer.value.decode("utf-8"), error_code, samps_per_chan_written)
nidaqmx.errors.DaqWriteError: A hardware failure has occurred. The operation could not be completed as specified.
Task Name: _unnamedTask<0>

Status Code: -50152
```

When I comment out the cfg_samp_clk_timing line (i.e. not using hardware timing), I do not get the error message:

```
1 Channel N Samples Write: 
5
```

How can I troubleshoot this? I am a nidaq newbie, so it could be something very simple.

bkeryan, 2023-06-20:
Also, does the `dmesg` command display any DMA errors? See "DMA is not working with NI Drivers on IOMMU-enabled systems" for more info: https://www.ni.com/en-us/support/documentation/bugs/23/ni-linux-device-drivers-2023-q2-known-issues.html

williamsnider, 2023-06-22:
Output of `dmesg | grep "ni"`  (let me know if I should run something else)
```
[    0.000000] random: crng init done
[    0.003664] check: Scanning 1 areas for low memory corruption
[    0.004331] Initmem setup node 0 [mem 0x0000000000001000-0x000000042f2fffff]
[    0.019012] mem auto-init: stack:off, heap alloc:on, heap free:off
[    0.036844] Memory: 16230404K/16684788K available (16393K kernel code, 3934K rwdata, 10196K rodata, 2608K init, 5604K bss, 454124K reserved, 0K cma-reserved)
[    0.385810] pid_max: default: 32768 minimum: 301
[    0.385829] LSM: Security Framework initializing
[    0.385846] AppArmor: AppArmor initialized
[    0.386267] Spectre V1 : Mitigation: usercopy/swapgs barriers and __user pointer sanitization
[    0.551290] devtmpfs: initialized
[    0.551290] pinctrl core: initialized pinctrl subsystem
[    0.551290] audit: initializing netlink subsys (disabled)
[    0.553819] audit: type=2000 audit(1687458038.192:1): state=initialized audit_enabled=0 res=1
[    0.591034] ACPI: \_SB_.PCI0.SBRG.EC0_: Boot DSDT EC initialization complete
[    0.591065] SCSI subsystem initialized
[    0.591065] NetLabel: Initializing
[    0.606558] pnp: PnP ACPI init
[    0.615559] Trying to unpack rootfs image as initramfs...
[    0.681206] Freeing initrd memory: 52040K
[    0.687833] check: Scanning for low memory corruption every 60 seconds
[    0.688313] Initialise system trusted keyrings
[    0.689219] fuse: init (API version 7.32)
[    0.689306] integrity: Platform Keyring initialized
[    0.695649] Monitor-Mwait will be used to enter C-1 state
[    0.812299] tun: Universal TUN/TAP device driver, 1.6
[    0.812425] uhci_hcd: USB Universal Host Controller Interface driver
[    0.812835] device-mapper: ioctl: 4.43.0-ioctl (2020-10-01) initialised: dm-devel@redhat.com
[    0.812903] drop_monitor: Initializing network drop monitor service
[    0.817667] resctrl: L3 monitoring detected
[    0.818343] Key type fscrypt-provisioning registered
[    0.819698] evm: Initialising EVM extended attributes:
[    0.887484] RAS: Correctable Errors collector initialized.
[    0.888523] Freeing unused kernel image (initmem) memory: 2608K
[    0.920190] Run /init as init process
[    0.920191]     /init
[    1.167769] e1000e 0000:05:00.0 0000:05:00.0 (uninitialized): registered PHC clock
[    5.086510] systemd[1]: systemd 245.4-4ubuntu3.20 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD +IDN2 -IDN +PCRE2 default-hierarchy=hybrid)
[    5.228530] systemd[1]: Listening on RPCbind Server Activation Socket.
[    5.228605] systemd[1]: Listening on Syslog Socket.
[    5.228648] systemd[1]: Listening on fsck to fsckd communication Socket.
[    5.228671] systemd[1]: Listening on initctl Compatibility Named Pipe.
[    5.228746] systemd[1]: Listening on Journal Audit Socket.
[    5.228786] systemd[1]: Listening on Journal Socket (/dev/log).
[    5.228833] systemd[1]: Listening on Journal Socket.
[    5.229092] systemd[1]: Listening on udev Control Socket.
[    5.229124] systemd[1]: Listening on udev Kernel Socket.
[    5.242378] systemd[1]: Finished Create list of static device nodes for the current kernel.
[    5.243735] systemd[1]: Finished Load Kernel Module pstore_blk.
[    5.244214] systemd[1]: Finished Load Kernel Module pstore_zone.
[    5.244610] systemd[1]: Finished Uncomplicated firewall.
[    5.248077] systemd[1]: Finished Load Kernel Module efi_pstore.
[    5.249451] systemd[1]: Finished Load Kernel Module drm.
[    5.250249] systemd[1]: Finished Load Kernel Module ramoops.
[    5.250770] systemd[1]: Finished Remount Root and Kernel File Systems.
[    5.261132] systemd[1]: Finished Create System Users.
[    5.262744] systemd[1]: Finished Set the console keyboard layout.
[    5.264157] systemd[1]: Finished Load Kernel Modules.
[    5.277795] systemd[1]: Finished Load/Save Random Seed.
[    5.282526] systemd[1]: Finished Create Static Device Nodes in /dev.
[    5.284833] systemd[1]: Mounting Mount unit for bare, revision 5...
[    5.287360] systemd[1]: Mounting Mount unit for code, revision 131...
[    5.290386] systemd[1]: Mounting Mount unit for code, revision 132...
[    5.293927] systemd[1]: Mounting Mount unit for core, revision 14946...
[    5.298308] systemd[1]: Mounting Mount unit for core, revision 15419...
[    5.301408] systemd[1]: Mounting Mount unit for core18, revision 2751...
[    5.305340] systemd[1]: Mounting Mount unit for core18, revision 2785...
[    5.307404] systemd[1]: Mounting Mount unit for core20, revision 1891...
[    5.310927] systemd[1]: Mounting Mount unit for core20, revision 1950...
[... 46 lines cut ...]
[ 1075.812042] niraptnonusb 0000:04:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0017 address=0x107942000 flags=0x0000]
[ 1075.812049] niraptnonusb 0000:04:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0017 address=0x138204000 flags=0x0020]
[ 1087.679601] niraptnonusb 0000:04:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0017 address=0x107942000 flags=0x0000]
[ 1087.679608] niraptnonusb 0000:04:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0017 address=0x138204000 flags=0x0020]
[ 1118.442819] niraptnonusb 0000:04:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0017 address=0x107942000 flags=0x0000]
[ 1118.442826] niraptnonusb 0000:04:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0017 address=0x138204000 flags=0x0020]
[... cut ...]
[ 5136.044486] niraptnonusb 0000:04:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0017 address=0x10a72a000 flags=0x0000]
[ 5136.044493] niraptnonusb 0000:04:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0017 address=0x138204000 flags=0x0020]
[ 7996.375646] niraptnonusb 0000:04:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0017 address=0x107942000 flags=0x0000]
[ 7996.375653] niraptnonusb 0000:04:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0017 address=0x138204000 flags=0x0020]
```

williamsnider, 2023-06-22:
I also tried a different nidaq board (PCIe-6251) on a different PC (also running Ubuntu 20.04). I followed the steps here (https://www.ni.com/docs/en-US/bundle/ni-platform-on-linux-desktop/page/installing-ni-products-ubuntu.html) to install the linux 2023Q2 drivers. On this new setup, when I run this code, I get a different error:

zhindes, 2023-06-22:
@williamsnider - The `dmesg` log looks to show the smoking gun. `niraptnonusb` is the kernel module used by NI-DAQmx to communicate with an X Series device. `AMD-Vi` is "AMD I/O Virtualization Technology (IOMMU)". Brad's link to the [known issues](https://www.ni.com/en/support/documentation/bugs/23/ni-linux-device-drivers-2023-q2-known-issues.html) mentions a workaround.

> Disable IOMMU control from the Linux kernel bootup parameters with whichever of these applies to the system:
> iommu=off  
> intel_iommu=off 
> amd_iommu=off 

Are you in a situation where you are able to try those to see if the problem is resolved?

closed 2023-06-22 by williamsnider