ni/nidaqmx-python #424: How to update data using a stream_writer in a hardware-timed, synchronous read/write setup?

williamsnider, 2023-06-27:
Hello, this is almost certainly not an issue with the python bindings, but I've tried to follow the examples and documentation as well as different posts online to no avail, so I am posting here for help.

For my application, I want to have hardware-timed, synchronous read/write digital signals using a PCIe-6321. I'm using a stream_writer and stream_reader for the performance boost. However, I am not sure how to update the value that is written by the stream_writer. Here is my code:

```
import nidaqmx
from nidaqmx.constants import LineGrouping, AcquisitionType, RegenerationMode
import numpy as np
import time
from nidaqmx.stream_readers import DigitalSingleChannelReader
from nidaqmx.stream_writers import DigitalSingleChannelWriter

SAMPLE_RATE = 1000000  # hz
SAMPLES_PER_CHANNEL = 100
SAMPLES_FOR_DATA = 10

def generate_random_data():
    """Generate random boolean data with size SAMPLES_PER_CHANNEL"""
    data_write = np.random.choice([0, 2], SAMPLES_PER_CHANNEL).astype(
        "uint8"
    )  # 2 because writing to line1?
    data_write[-1] = 0  # Always end LOW
    return data_write

with nidaqmx.Task() as read_task, nidaqmx.Task() as write_task:

    # Add virtual channels
    # Note - line0 and line1 are physically connected in my setup
    read_task.di_channels.add_di_chan(
        "Dev2/port0/line0", line_grouping=LineGrouping.CHAN_PER_LINE
    )
    write_task.do_channels.add_do_chan(
        "Dev2/port0/line1", line_grouping=LineGrouping.CHAN_PER_LINE
    )

    # Set hardware timing
    read_task.timing.cfg_samp_clk_timing(
        SAMPLE_RATE,
        sample_mode=AcquisitionType.FINITE,
        samps_per_chan=SAMPLES_PER_CHANNEL,
    )
    write_task.timing.cfg_samp_clk_timing(
        SAMPLE_RATE,
        sample_mode=AcquisitionType.FINITE,
        samps_per_chan=SAMPLES_PER_CHANNEL,
    )

    # Turn off regeneration
    write_task.out_stream.regen_mode = RegenerationMode.DONT_ALLOW_REGENERATION

    # Create numpy arrays for streams
    data_read = np.zeros(SAMPLES_FOR_DATA, dtype="uint8")
    data_write = generate_random_data()

    # Streams for faster read/write
    reader = DigitalSingleChannelReader(read_task.in_stream)
    writer = DigitalSingleChannelWriter(write_task.out_stream, auto_start=False)

    # Trigger write_task once read_task is started
    write_task.triggers.start_trigger.cfg_dig_edge_start_trig(
        read_task.triggers.start_trigger.term
    )

    # Initial data for write_task
    writer.write_many_sample_port_byte(data_write)
    write_task.start()  # Data not written until triggered by read_task

    for _ in range(3):

        start = time.time()
        reader.read_many_sample_port_byte(
            data_read, number_of_samples_per_channel=SAMPLES_FOR_DATA
        )
        stop = time.time()

        print("-" * 10)
        print(round(stop - start, 6), "s to read/write data")
        print(
            data_write[:SAMPLES_FOR_DATA],
            " data written (truncated to first 10 values).",
        )
        print(data_read, " data read.")

        # Update data_write for next iteration of loop  -- this is not working
        data_write = generate_random_data()

    write_task.stop()
    read_task.stop()
```

The output I get is:
```
----------
0.002054 s to read/write data
[0 2 2 2 2 0 2 2 0 2]  data written (truncated to first 10 values).
[0 0 1 1 1 1 0 1 1 0]  data read.
----------
6.2e-05 s to read/write data
[0 0 0 2 0 2 0 2 2 2]  data written (truncated to first 10 values).
[1 0 0 0 1 1 0 1 1 1]  data read.
----------
5.5e-05 s to read/write data
[2 0 2 2 2 0 0 0 0 2]  data written (truncated to first 10 values).
[0 1 0 0 1 1 1 0 0 1]  data read.
```
For the first iteration of the loop, the data read is the same as the data written (with a 1 sample offset -- this is fine). However, on subsequent iterations of the loop, the data read differs from the data "written". How can I update the values that should be written and still have the read_task adequately trigger the write_task?

bkeryan, 2023-06-27:
Modifying `data_write` has no effect because DAQmx doesn't save a reference to `data_write`. Instead, `write_task` has a circular output buffer that is managed by the DAQmx driver. 

[Continuous Acquisition and Generation with Finite Buffer Size](https://www.ni.com/docs/en-US/bundle/ni-daqmx/page/mxcncpts/contcacqgen.html) has a diagram:
[image]

Calling `writer.write_many_sample_port_byte()` copies `data_write` into `write_task`'s circular output buffer. When you modify `data_write`, you need to call `writer.write_many_sample_port_byte()` again for the changes to take effect.

When do the changes take effect? That depends.
- By default, `write_task` [regenerates samples](https://www.ni.com/docs/en-US/bundle/ni-daqmx/page/mxcncpts/regen.html), which means it repeatedly outputs the samples that you wrote in the first call to `writer.write_many_sample_port_byte()`. You may want to disable regeneration to ensure that `write_task` always generates newly written data.
- The first call to `writer.write_many_sample_port_byte()` also sets the output buffer size. If you write again with the same size, then you are updating the entire output buffer, which can cause slow updates because DAQmx is almost always trying to read the part of the buffer that you're updating. It's often better to use double buffering, where the first write is 2x the size of subsequent writes. That way, subsequent writes only update half of the circular output buffer at a time.

Other topics in [NI-DAQmx Key Concepts >> Reading and Writing Data >> Buffering](https://www.ni.com/docs/en-US/bundle/ni-daqmx/page/mxcncpts/buffering.html) may also be relevant.

williamsnider, 2023-06-30:
This is very helpful information - thank you!

closed 2023-06-30 by williamsnider