ni/nidaqmx-python #432: Sample callback triggered more often than expected

aranvir, 2023-07-06:
## Description

Hi! I have observed that for sample rates lower than 1 kHz, more data is generated than expected. For example, if I record for 5 seconds at a rate of 100 Hz, I expect 500 samples to be produced. Instead, 5000 samples are produced. Interestingly, this is inverse proportional to the ratio of `sample_frequency/1kHz`. Ergo, if I record with 500 Hz I expect 2000 samples. At 1 kHz and above, the sample size is always `sample_time * sample_frequency`.

I varied the `samples_per_channel` parameter but it did not have a an effect on the problem at hand.

To my understanding, the `reading_task_callback()` should be triggered every time `samples_per_channel` can be read from the buffer. When I set the frequency to 1000 Hz, the sample_channels to 1000 and record for 2 seconds, I see exactly 2 callback events before the loop stops. For 100 Hz and sample_channels = 100, I would expect the same pattern, but instead I see 10 events.

So what could be the cause of this? I could not identify a "lower sampling frequency limit" for the hardware, so I'm wondering if it's a configuration issue on software side somehow...

## Versions

Windows 10
Python 3.9.13
Nidaqmx 0.7.0
NI Version 2022 Q3

Acquisition unit: NI cDAQ-9174 with NI 9232
Sensor: PCB Piezotronics M350C02 Shock Accelerometer

## Code used

This is the code I use for my samples. The sample frequency is `self.sample_rate` and the channels per sample are `self.channel_samples`.

```python
    def start_monitoring(self):
        self.buffer = np.zeros((self.num_sensors, self.channel_samples), dtype=np.float64)
        self.task = nidaqmx.Task()
        self.task.add_ai_accel_chan("cDAQ1Mod1/ai0")
        self.task.timing.cfg_samp_clk_timing(self.sample_rate, sample_mode=constants.AcquisitionType.CONTINUOUS,
                                             samps_per_chan=self.channel_samples)
        reader = stream_readers.AnalogMultiChannelReader(self.task.in_stream)

        def reading_task_callback(task_idx, event_type, num_samples, callback_data=None):  # pylint: disable=W0613
            buffer = np.zeros((self.num_sensors, num_samples), dtype=np.float64)
            logging.debug("Samples collected: %s", num_samples)
            reader.read_many_sample(buffer, num_samples, timeout=constants.WAIT_INFINITELY)
            self.acquired_data.append(buffer)
            if time.time() - self.time > self.timeout:
                self.stop_recording.emit()  # qt-signal for timed recording stop
            return 0

        self.task.register_every_n_samples_acquired_into_buffer_event(self.channel_samples, reading_task_callback)
        self.time = time.time()
        self.task.start()
```

bkeryan, 2023-07-10:
> I could not identify a "lower sampling frequency limit" for the hardware,

https://www.ni.com/docs/en-US/bundle/ni-9232-specs/page/specs.html says 
```
Data rate range ( fs) using internal master timebase
--
Minimum | 0.985 kS/s
Maximum | 102.4 kS/s
```

I think [Valid Sampling Rates for NI DSA, CM C Series and SC Express Devices
 ](https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z000000P6FrSAK&l=en-US) describes the behavior you're seeing: 100 Hz is lower than the minimum supported rate, so DAQmx rounds up to the closest supported rate, which is 985 Hz.
 
 You can read back the actual sample rate by getting the `task.timing.samp_clk_rate` property.

closed 2023-07-10 by zhindes

aranvir, 2023-07-11:
@bkeryan thanks a lot!! That's exactly what happens.