ni/nidaqmx-python #447: Error in the example "do_sw_timed.py"

QDO164, 2023-08-02:
An error occurs in execution of the example "do_sw_timed.py"

"Write cannot be performed when the auto start input to DAQmx Write is false, task is not running, and timing for the task is not configured or Timing Type is set to On Demand.
Set auto start to true, start the task, or configure timing and specify Timing Type other than On Demand.
Task Name: _unnamedTask<0>
Status Code: -200846"

A Boolean list is not accepted.
NG: task.write([True, False, True, False])
A decimal number or Boolean is accepted.
OK: task.write(8)
OK: task.write(True)

python 3.11.3, nidaqmx 0.8.0, numpy 1.24.3, NI-DAQmx 2023 Q3, Windows 10, NI USB-6218

bkeryan, 2023-08-02:
@QDO164, thanks for reporting this. I think the "error expected" code is confusing and should be removed from the example. Examples should demonstrate what to do, not what not to do.

QDO164, 2023-08-08:
Thank you, @bkeryan.

bkeryan, 2023-08-08:
I think we should keep this issue open until this code is removed from the example.