ni/nidaqmx-python #454: Analog Output providing unexpected output

rfebbo, 2023-10-05:
I have a PXIe-6365. I am working with the hw_timed ao example with the following code:

```
with nidaqmx.Task() as task:
    task.ao_channels.add_ao_voltage_chan("PXI1Slot4/ao0")

    task.timing.cfg_samp_clk_timing(1000)

    print("1 Channel N Samples Write: ")
    print(task.write([1.1, 2.2, 3.3, 4.4, 5.5], auto_start=False))
    task.start()
    task.wait_until_done()
    
    task.stop()
```

When I measure the output on my oscilloscope I get the 5 different voltages but they are repeated over the course of a second. I was expecting only 5 voltage shifts in 1 ms increments over 5 ms but I am getting more like 1000 voltage shifts over a second.

[image]

[image]

Is this expected? If so, what should I change to get only 5 voltage shifts in 1 ms increments over 5 ms?

Thanks

bkeryan, 2023-10-05:
The `cfg_samp_clk_timing` method takes an optional `samps_per_chan` parameter ([docs](https://nidaqmx-python.readthedocs.io/en/latest/timing.html#nidaqmx._task_modules.timing.Timing.cfg_samp_clk_timing)) that defaults to 1000 samples per channel. Try specifying `task.timing.cfg_samp_clk_timing(1000, samps_per_chan=5)`.

Also, reading `task.out_stream.total_samp_per_chan_generated` ([docs](https://nidaqmx-python.readthedocs.io/en/latest/out_stream.html#nidaqmx._task_modules.out_stream.OutStream.total_samp_per_chan_generated)) can be helpful for debugging issues like this.

zhindes, 2023-10-05:
Its worth noting that @bkeryan's suggestion works because `cfg_samp_clk_timing` has another default parameter, `sample_mode=AcquisitionType.FINITE`, which ensures the generation will stop once all the samples have been generated. If you set that to `AcquisitionType.CONTINUOUS` you would see similar behavior from your screenshot, but into infinity since `task.wait_until_done()` would never return.

zhindes, 2023-10-05:
Closing as its not a Python binding specific problem, and we've gotten them on the right track.

closed 2023-10-05 by zhindes

rfebbo, 2023-10-05:
This resolved my issue, thank you very much for the feedback and links.