ni/nidaqmx-python #456: How to read TEDS information from sensor

sekisushai, 2023-10-19:
Hello,
I would like to display the information of a TEDS sensor (microphone) using nidaqmx-python.
I did not find any method to to that ? Is it possible ? How to do it ?
Thank you,
Pierre

zhindes, 2023-10-29:
We don't typically expose reading the TEDS values directly, but you see the datasheet in MAX. Instead, we eliminate configuration parameters that we ready from the datasheet. So, for example, you can see that we don't ask you for `mic_sensitivity`.

```python
    def add_ai_microphone_chan(
            self, physical_channel, name_to_assign_to_channel="",
            terminal_config=TerminalConfiguration.DEFAULT,
            units=SoundPressureUnits.PA, mic_sensitivity=10.0,
            max_snd_press_level=100.0,
            current_excit_source=ExcitationSource.INTERNAL,
            current_excit_val=0.004, custom_scale_name=""):

    def add_teds_ai_microphone_chan(
            self, physical_channel, name_to_assign_to_channel="",
            terminal_config=TerminalConfiguration.DEFAULT,
            units=SoundPressureUnits.PA, max_snd_press_level=100.0,
            current_excit_source=ExcitationSource.INTERNAL,
            current_excit_val=0.004, custom_scale_name=""):
```

closed 2023-10-29 by zhindes

bkeryan, 2023-10-29:
BTW, NI provides a LabVIEW library for parsing IEEE 1451.4 TEDS datasheets: https://www.ni.com/en/support/documentation/supplemental/06/ieee-1451-4-sensor-templates-overview.html

I'm not aware of a Python library for parsing TEDS datasheets.

DAQmx's TEDS parsing support is built into the DAQmx DLLs and is focused on translating the datasheet to DAQmx channel properties, not providing the datasheet values to users.

bkeryan, 2023-10-29:
Also: if you create a `nidaqmx.Task` and add a microphone channel to it, you should be able to read back the [ai_microphone_sensitivity](https://nidaqmx-python.readthedocs.io/en/latest/ai_channel.html#nidaqmx._task_modules.channels.ai_channel.AIChannel.ai_microphone_sensitivity) channel property.