ni/nidaqmx-python #457: Synchronization of two tasks

roflmaostc, 2023-11-07:
Hi,

we were trying to create two output signals where one of the frequencies is 2 times slower than the other one.

We tried the following:
```python

import nidaqmx
from nidaqmx.constants import TriggerType, Edge, AcquisitionType, TaskMode, LineGrouping
import nidaqmx.system

from PyDAQmx.DAQmxFunctions import *
from PyDAQmx.DAQmxConstants import *

class ContinuousPulseTrainGeneration():
    """ Class to create a continuous pulse train on a counter
    
    Usage:  pulse = ContinuousTrainGeneration(period [s],
                duty_cycle (default = 0.5), counter (default = "dev1/ctr0"),
                reset = True/False)
            pulse.start()
            pulse.stop()
            pulse.clear()
    """
    def __init__(self, period=1., duty_cycle=0.5, counter="Dev1/ctr0", reset=False):
        if reset:
            DAQmxResetDevice(counter.split('/')[0])
        taskHandle = TaskHandle(0)
        DAQmxCreateTask("",byref(taskHandle))
        DAQmxCreateCOPulseChanFreq(taskHandle,counter,"",DAQmx_Val_Hz,DAQmx_Val_Low,
                                                                   0.0,1/float(period),duty_cycle)
        DAQmxCfgImplicitTiming(taskHandle,DAQmx_Val_ContSamps,1000)
        self.taskHandle = taskHandle
    def start(self):
        DAQmxStartTask(self.taskHandle)
    def stop(self):
        DAQmxStopTask(self.taskHandle)
    def clear(self):
        DAQmxClearTask(self.taskHandle)

# initialize DAQ
print("Initializing NI DAQ...")
system = nidaqmx.system.System.local()
system.driver_version
for device in system.devices:
    print(device)
          

daq = system.devices['Dev1']
daq == nidaqmx.system.Device('Dev1')

# DEFINE DAQ LINES
dmd_trig_line = "Dev1/ctr3"
camera_trig_line = "Dev1/ctr2"

dmd_trig_period =  0.0005
pulse_DMD = ContinuousPulseTrainGeneration(period=dmd_trig_period, 
                                           duty_cycle=0.5,
                                           counter=dmd_trig_line, 
                                           reset=True)

pulse_CAMERA = ContinuousPulseTrainGeneration(period=2 * dmd_trig_period, 
                                           duty_cycle=0.5,
                                           counter=camera_trig_line, 
                                           reset=True)

pulse_DMD.start()
pulse_CAMERA.start()

```

However, on the oscilloscope we could see that the two signals are not well aligned in time. It seems like there is the synchronization missing. 

We tried with `DAQmxCfgSampClkTiming` to get a `clock_task_handle` but it didn't work because of weird errors (`DaqError: Specified operation cannot be performed when there are no devices in the task.
Task Name: _unnamedTask<0>`)

Can you point us to any example how to synchronize two different task such they are well aligned in time?

Thanks a lot,

Felix

zhindes, 2023-11-08:
You are mixing two different APIs. [PyDAQmx](https://pypi.org/project/PyDAQmx/) is a separate project from [nidaqmx](https://pypi.org/project/nidaqmx/), which is open-sourced in this repository. `PyDAQmx` is basically a straight wrapper of the NI-DAQmx C API. `nidaqmx` is implemented on top of the same C API, but is much more Pythonic, with classes and abstractions for most DAQmx concepts. To implement something similar to what you have done by using `nidaqmx` check out the [co_pulse_time.py](https://github.com/ni/nidaqmx-python/blob/master/examples/co_pulse_time.py) example. I'll inline it here, since its simple.

```python
with nidaqmx.Task() as task:
    task.co_channels.add_co_pulse_chan_time("Dev1/ctr1", low_time=0.01, high_time=0.01)
    task.timing.cfg_implicit_timing(sample_mode=AcquisitionType.CONTINUOUS)
    task.start()

    print("Waiting before changing pulse specification...")
    time.sleep(2)

    sample = CtrTime(high_time=0.001, low_time=0.002)

    print("1 Channel 1 Sample Write: ")
    print(task.write(sample))
```

The only major differences are that example uses Time instead of Frequency and also updates the pulse specification after a short while. You should be able to modify those characteristics.

> However, on the oscilloscope we could see that the two signals are not well aligned in time. It seems like there is the synchronization missing.

Correct. You need to ensure they start at the same time. You need to configure a [Start Trigger](https://nidaqmx-python.readthedocs.io/en/latest/start_trigger.html#). Something like this:

```python
task2.triggers.start_trigger.cfg_dig_edge_start_trig(task1.triggers.start_trigger.term)
# ...
task2.start()
task1.start()
```

That will share the first task's Start Trigger with the second Task. Then if you start the second task first (😵 - sorry for the confusing terminology), then when the first task starts, the second task should also start at the same time (within a couple of ticks of the hardware timebase).

_Note: I don't have hardware handy to test this. I believe this simple implicit counter pulse generation should support the sharing of triggers like I just showed. If not, then you may need to use the `arm_start_trigger`, which I don't think has the `term` property. Hopefully you can piece that together._

closed 2023-11-08 by zhindes

roflmaostc, 2023-12-13:
I forgot to thank for your answer :)!

We didn't test yet but we will soon !