ni/nidaqmx-python #484: Update of NI-DAQ without a reboot causes SIGABRT on the application

pieleric, 2024-02-02:
Hi,
After every update of the NI packages (on Ubuntu), if I don't restart the computer, using nidaqmx in my application automatically causes a SIGABRT (6) and kills the whole application. The only "information" is confusing error message:
```
libnipalu.so failed to initialize
Verify that nipalk.ko is built and loaded.
```
That's confusing because `nipalk.ko` is loaded... just an older version than the NI-DAQ library. It's annoying because our users don't typically run our application in a terminal, so they don't see any error message. Also, as the whole process is killed, it's impossible to show a nicer error message, or to handle it.

Is it possible to handle such case more gracefully and **raise an exception when loading nidaqmx, instead of killing the process**?

For info here is the full backtrace when this happens:
```
[2818737] *** Process received signal ***
[2818737] Signal: Aborted (6)
[2818737] Signal code:  (-6)
[2818737] [ 0] /usr/lib/x86_64-linux-gnu/libc.so.6(+0x43090)[0x7f4d924c7090]
[2818737] [ 1] /usr/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb)[0x7f4d924c700b]
[2818737] [ 2] /usr/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b)[0x7f4d924a6859]
[2818737] [ 3] /usr/lib/x86_64-linux-gnu/libnipalu.so.1(+0x76836)[0x7f4d71b7b836]
[2818737] [ 4] /usr/lib/x86_64-linux-gnu/libnipalu.so.1(+0x25c00)[0x7f4d71b2ac00]
[2818737] [ 5] /lib64/ld-linux-x86-64.so.2(+0x11b9a)[0x7f4d926abb9a]
[2818737] [ 6] /lib64/ld-linux-x86-64.so.2(+0x11ca1)[0x7f4d926abca1]
[2818737] [ 7] /usr/lib/x86_64-linux-gnu/libc.so.6(_dl_catch_exception+0xe5)[0x7f4d925e4985]
[2818737] [ 8] /lib64/ld-linux-x86-64.so.2(+0x160cf)[0x7f4d926b00cf]
[2818737] [ 9] /usr/lib/x86_64-linux-gnu/libc.so.6(_dl_catch_exception+0x88)[0x7f4d925e4928]
[2818737] [10] /lib64/ld-linux-x86-64.so.2(+0x1560a)[0x7f4d926af60a]
[2818737] [11] /usr/lib/x86_64-linux-gnu/libdl.so.2(+0x134c)[0x7f4d9245c34c]
[2818737] [12] /usr/lib/x86_64-linux-gnu/libc.so.6(_dl_catch_exception+0x88)[0x7f4d925e4928]
[2818737] [13] /usr/lib/x86_64-linux-gnu/libc.so.6(_dl_catch_error+0x33)[0x7f4d925e49f3]
[2818737] [14] /usr/lib/x86_64-linux-gnu/libdl.so.2(+0x1b59)[0x7f4d9245cb59]
[2818737] [15] /usr/lib/x86_64-linux-gnu/libdl.so.2(dlopen+0x4a)[0x7f4d9245c3da]
[2818737] [16] /usr/lib/python3.8/lib-dynload/_ctypes.cpython-38-x86_64-linux-gnu.so(+0x15ac1)[0x7f4d90ccdac1]
[2818737] [17] python(PyCFunction_Call+0xfa)[0x5f69da]
[2818737] [18] python(_PyObject_MakeTpCall+0x296)[0x5f7506]
[2818737] [19] python(_PyEval_EvalFrameDefault+0x59c7)[0x570787]
[2818737] [20] python(_PyEval_EvalCodeWithName+0x26a)[0x5697da]
[2818737] [21] python[0x59c396]
[2818737] [22] python(_PyObject_MakeTpCall+0x1ff)[0x5f746f]
[2818737] [23] python(_PyEval_EvalFrameDefault+0x5dce)[0x570b8e]
[2818737] [24] python[0x50b07e]
[2818737] [25] python(_PyEval_EvalFrameDefault+0x5796)[0x570556]
[2818737] [26] python(_PyFunction_Vectorcall+0x1b6)[0x5f6ce6]
[2818737] [27] python(_PyEval_EvalFrameDefault+0x859)[0x56b619]
[2818737] [28] python[0x50343f]
[2818737] [29] python(PyObject_GetAttr+0x267)[0x5c2287]
[2818737] *** End of error message ***
Aborted (core dumped)
```
Our system: Ubuntu 20.04/22.04 64bits x86, with the latest version of nidaqmx-python.

bkeryan, 2024-02-02:
Your application may be able to display a nicer error message by handling SIGABRT: https://discuss.python.org/t/how-can-i-handle-sigabrt-from-third-party-c-code-std-abort-call/22078

I doubt that it will be effective to translate SIGABRT into a Python exception, because Linux will re-raise SIGABRT when the signal handler returns.

zhindes, 2024-02-02:
>  After every update of the NI packages (on Ubuntu), if I don't restart the computer, using nidaqmx in my application automatically causes a SIGABRT (6) and kills the whole application. The only "information" is confusing error message:

Does it not ask you to reboot? Anyways, I agree that handling this specific issue within the Python module isn't a very useful thing to do. If you have some great idea for that and want to contribute, we can take a look. In the mean-time it would be more effective to post this to either the [Multifunction DAQ Community on ni.com](https://forums.ni.com/t5/Multifunction-DAQ/bd-p/250) or to contact NI tech support.

closed 2024-02-02 by zhindes

pieleric, 2024-02-05:
Thanks @bkeryan , I also had a look at catching the SIGABRT from our Python code, but it's pretty hard to do anything with it meaningful as the process is still killed after "handling" the signal. I'll fallback to my second plan, which is similar to what is described in the thread you linked to: have some sort of "canary" process which just load nidaqmx, opens the device and quits. That should be sufficient to detect whether opening a NI device works or not.

@zhindes , no, the updater doesn't show any (graphical) message. I'll send a request on the forum to handle updates more "gracefully" on Debian/Ubuntu.

pieleric, 2024-02-06:
For info, I've now made a "canary" program that gets killed if the NI-DAQmx cannot run. It looks like that:
```
        canary_cmd = [sys.executable, "-c", "import nidaqmx; nidaqmx.system.System.local().devices"]
        process = subprocess.run(canary_cmd)
        if process.returncode < 0:
            raise model.IOError("NI-DAQmx failed to load, reboot the computer and try again.")
```