ni/nidaqmx-python #543: Constant and inconstant values using NI-DAQmx Python API although not issues with NI SignalExpress 2015

bjce, 2024-03-18:
**Introduction**

I encountered issues with extracting data using NI-DAQmx Python algorithms (using Python 3.11 on Anaconda, with Spyder as the IDE). The data remains unchanged when I alter the settings, yielding a constant, incorrect value. This question outlines my troubleshooting process and the steps taken to resolve the issue.

**Checking the Device Connected to the National Instruments (NI-DAQmx)**

***MAX***

Initially, I utilized MAX (Measurement and Automation Explorer) to verify the device's functionality. The device appeared to operate correctly, and test panel readings matched expected theoretical values :

[image]

***NI SignalExpress 2015***

Further testing with NI SignalExpress 2015 confirmed that the software could accurately retrieve the expected values, indicating no issues at this stage.

[image]

**Code Implementation in NI-DAQmx Python**

Despite anticipating seamless integration with the NI-DAQmx Python API, problems emerged: The values returned by basic algorithms were static and inconsistent with both expected fluctuations and previous measurements, displaying values such as 5.3, which were incongruent with prior results. Below, I detailed the algorithms attempted and their respective outcomes.

***First Algorithms and Results***

[image]

[image]

***Second Algorithms and Results***

[image]

**Bug Resolution attempt**

I verified the device name and ports, ensuring accuracy in their specification. Additionally, I regularly cleared the device of any residual data to prevent issues related to data overflow or corruption.

**Conclusion**

I am at a loss regarding the cause of these discrepancies between NI SignalExpress 2015 and NI-DAQmx or how to rectify them. Any assistance or guidance would be greatly appreciated.

bkeryan, 2024-03-18:
Hi @bjce,

In MAX and SignalExpression, you are acquiring data with a +/-10 V range, but the Python `add_ai_voltage_channel` method sets `min_val=-5.0` and `max_val=5.0` by default:
```
    def add_ai_voltage_chan(
            self, physical_channel, name_to_assign_to_channel="",
            terminal_config=TerminalConfiguration.DEFAULT, min_val=-5.0,
            max_val=5.0, units=VoltageUnits.VOLTS, custom_scale_name=""):
```

This is clipping your data to approximately a +/-5 V range. The reason it reads 5.3 V rather than 5 V is that X Series devices have a small amount of overrange so that software calibration doesn't reduce the usable range.

Try specifying `min_val=-10.0` and `max_val=10.0`.

bjce, 2024-03-19:
Many many thanks @bkeryan for your time and your help. It worked perfefctly

Best regards

closed 2024-03-19 by bkeryan