ni/nidaqmx-python #602: types.py makes local debugging difficult

mshafer-NI, 2024-06-20:
When attempting to run a single script, the cwd (which is vscode will be the folder of the script) is considered for imports first. This causes Python to import ./types when expecting to import stdlib types (https://docs.python.org/3/library/types.html).

This violates [[O.2.1] ❌ DO NOT Redefine or “shadow” declarations](https://ni.github.io/python-styleguide/#O-2-1)

bkeryan, 2024-06-20:
@mshafer-NI Does this affect users of nidaqmx-python or does it only affect developers working on nidaqmx-python? For users, the current directory should be outside the `nidaqmx` package.

Renaming `nidaqmx.types` to something else would break compatibility. If this affects users, would it be acceptable to rename the submodule to `_types.py` and alias it in `nidaqmx/__init__.py`?
```python
import _types as types

__all__ = [..., "types"]
```

mshafer-NI, 2024-06-20:
It only affects developers (as clients are not expected to be working IN the library folder).

```python
import _types as types

__all__ = [..., "types"]
```

This is a good solution