ni/nidaqmx-python #619: Reading position from a quadrature encoder

sarahsfarrell, 2024-07-25:
I am using an NI cDAQ-9184 with the NI 9411 module to read a quadrature encoder, but I can't seem to figure out how to get position readings from it. I have a voltage pulse signal wired into the 9411 to act as my external sample clock, and the sample timestamps seem to be right, but I seem to be getting edge counts or something rather than actual angular position measurements. I'm assuming I need to use some other piece of the library to convert to position, but I can't seem to figure that part out. I would welcome any help!

task_a = nidaqmx.Task()
    sample_rate = 2000  # Hz
    n_tot = sample_rate*seconds_to_record
    data_matrix = np.zeros((n_tot,))

    task_a.ci_channels.add_ci_ang_encoder_chan(
        counter="cDAQ9184-1C3169FMod3/ctr0",
        name_to_assign_to_channel="encoder 1",
        decoding_type=nidaqmx.constants.EncoderType.X_1,
        zidx_enable=True,
        units=nidaqmx.constants.AngleUnits.DEGREES,
        pulses_per_rev=4096
    )

    task_a.timing.cfg_samp_clk_timing(
        rate=sample_rate,
        source="/cDAQ9184-1C3169FMod3/PFI3",  
        sample_mode=AcquisitionType.CONTINUOUS,
        samps_per_chan=sample_rate*seconds_to_record
    )

    print("beginning data collection. time of collection is approx ", seconds_to_record, " seconds.")

    task_a.start()
    stream = stream_readers.CounterReader(task_a.in_stream)

    stream.read_many_sample_double(
        data=data_matrix,
        number_of_samples_per_channel=n_tot
    )

    task_a.stop()
    task_a.close()

zhindes, 2024-07-25:
A few questions:
* What is the data you are reading?
* How have you hooked your encoder up to the 9411?
* Do you know the part number or have the datasheet for your encoder?

For details on how encoders work see [NI Hardware Encoder Measurements: How-To Guide](https://knowledge.ni.com/KnowledgeArticleDetails?id=kA03q000000x1riCAA&l=en-US).

My best guess at what might be happening is that your encoder connections to the 9411 don't match the DAQmx defaults. See [CIChannel.ci_encoder_a_input_term](https://nidaqmx-python.readthedocs.io/en/latest/task_channels.html#nidaqmx.task.channels.CIChannel.ci_encoder_a_input_term), [CIChannel.ci_encoder_b_input_term](https://nidaqmx-python.readthedocs.io/en/latest/task_channels.html#nidaqmx.task.channels.CIChannel.ci_encoder_b_input_term), and [CIChannel.ci_encoder_z_input_term](https://nidaqmx-python.readthedocs.io/en/latest/task_channels.html#nidaqmx.task.channels.CIChannel.ci_encoder_z_input_term)

sarahsfarrell, 2024-07-25:
The data I am reading looks like a series of extremely quick fluctuations when I turn the encoder about 45 degrees, which is what led me to assume I'm reading pulses instead of position. Every peak maxes around +/-0.09, the sign seemingly dependent on which direction it's rotating
[image]
zoomed in image: 
[image]

The encoder is hooked up using the AMT-18C-3-072 cord from the encoder manufacturer (ending in pigtails) wired into a DB15 connector, which plugs in directly to the 9411. I did follow the diagram from NI on which wires to connect where, but maybe the defaults are different in the nidaqmx package? 
[image]

The encoder is the AMT132Q-V from CUI devices, datasheet here: [CUI AMT Series Encoders](https://cdn.automationdirect.com/static/specs/amtencoders.pdf?_gl=1*ikhz2n*_up*MQ..&gclid=Cj0KCQjwxqayBhDFARIsAANWRnS1PwmC0RTyEpN595qxR-QGsRAyIVqE_01dGoEFbjIrMB-ucdmwPb4aAt9YEALw_wcB)

sarrfar, 2024-09-25:
I just came back to this after a long hiatus - setting the defaults seems to have fixed everything. I'm now getting numbers that make sense. Great call, thank you!
[image]