ni/nidaqmx-python #636: Signal Fluactions While Acquiring PMT Analog Input

RealBrandonChen, 2024-09-12:
Hi,

I meet an issue in my system that I use a PMT to detect the photon intensity in a time sequence and reconstruts 2D image with the raster-scanning method. The PMT is connected with the DAQ (NI USB-6240) but I observe the strips-like structure in my images. I've inspected my light source, the PMT and they're definitely no problem. The reason is from the DAQ as you can see in the spliced images and the intensity profile, when I have 366(distance between two strips, with pixels)*256(image pixels in one direction)*2(multi-sampling) = 187392 sampling points, there is a fluaction raised. The code for DAQ setting is:

`
 class daq_samp():
      dev_name = 'Dev1'
      task_ai = [] # DAQmx task handle for analog input

    def __init__(self, sample_rate, im_size, showplot, autosetup=True, savefig=False):
        # to eliminate the Gaussian noise, we acquire 10 or 2 samples per focused image,
        self.multisampling = 1
        self.sample_rate = sample_rate * self.multisampling
        self.im_size = im_size
        self.scanning_points = self.im_size**2 
        self.showplot = showplot
        self.savefig = savefig
        self.img_sum = []
        
        if autosetup:
            self.set_up_daq()
        if showplot:
            self.setup_plot()
            
    def set_up_daq(self):
        self.task_ai = nidaqmx.Task()
        self.task_ai.ai_channels.add_ai_voltage_chan('%s/ai0' % self.dev_name)
        self.task_ai.timing.cfg_samp_clk_timing(self.sample_rate, active_edge=Edge.RISING, samps_per_chan=self.scanning_points * self.multisampling,
                                                sample_mode=AcquisitionType.CONTINUOUS)
        self.task_ai.triggers.start_trigger.cfg_dig_edge_start_trig("/Dev1/PFI0", Edge.RISING)

        # NOTE: must explicitly set the input buffer so that it's a multiple
        # of the number of samples per frame. Setting the samples per channel 
        # (above) does not achieve this.
        self.task_ai.in_stream.input_buf_size = self.scanning_points * 2 * self.multisampling
        
        # * Register a a callback function to be run every N samples
        self.task_ai.register_every_n_samples_acquired_into_buffer_event(self.scanning_points * self.multisampling, self.read_and_display_last_frame)
        # self.task_ai.register_every_n_samples_acquired_into_buffer_event(self.scanning_points * self.multisampling, self.callback)

bkeryan, 2024-09-12:
Have you ruled out ground-loop noise? Is the shell of the BNC connection ground-referenced, and if so, is it referenced to the same ground as the PC to which the USB-6210 is connected? See [Connecting Analog Input Signals on 
USB-6210/6211/6212 Devices (page 59 in the USB-621x User Manual
)](https://www.ni.com/docs/en-US/bundle/usb-621x-manual/resource/usb-621x-manual.pdf#page=59) and [Field Wiring and Noise Considerations for Analog Signals](https://www.ni.com/en/shop/data-acquisition/measurement-fundamentals/field-wiring-and-noise-considerations-for-analog-signals.html) for more info about how to handle different grounding scenarios.

bkeryan, 2024-09-12:
Also, this is unlikely to be a Python-binding specific issue. For further help, I recommend posting on the [NI Multifunction DAQ forum](https://forums.ni.com/t5/Multifunction-DAQ/bd-p/250) or contacting [NI support](https://www.ni.com/en/support.html).

closed 2024-09-12 by bkeryan