ni/nidaqmx-python #637: Repeatedly starting and stopping analog input task with registered callback causes BSOD (NI USB 6001)

dnzrgn, 2024-09-16:
Hello!

I've been having this problem for some time now, but I finally managed to recreate it reliably. Before, I would get it randomly, with multiple days without it, and then some days getting in multiple times. 

- I am using NI USB-6001 card. 
- OS is Windows 11 Pro, Version 10.0.22631 Build 22631
- NI-DAQmx version 2024 Q3.1 (though I recently updated while trying to fix the issue.)
- nidaqmx-python version 1.0.1 (was also experiencing problem with 0.9.0) 
- with python 3.11.7

Below code runs for a couple of seconds before windows crashes with a blue screen of death:

```
import nidaqmx
import nidaqmx.stream_readers
import numpy as np

device = nidaqmx.system.System().local().devices[0]

task = nidaqmx.Task("analog_input")
task.ai_channels.add_ai_voltage_chan(
    physical_channel=",".join([device.ai_physical_chans.channel_names[0], device.ai_physical_chans.channel_names[1]]),
    name_to_assign_to_channel="joystick",
    terminal_config=nidaqmx.constants.TerminalConfiguration.DIFF
)
task.ai_channels.add_ai_voltage_chan(
    physical_channel=device.ai_physical_chans[3].name,
    name_to_assign_to_channel="touch",
    terminal_config=nidaqmx.constants.TerminalConfiguration.RSE
)
task.timing.cfg_samp_clk_timing(rate=150, sample_mode=nidaqmx.constants.AcquisitionType.CONTINUOUS)

def _analog_samples_acquired_callback(task_handle, every_n_samples_event_type,
        number_of_samples, callback_data):
    try:
        analog_reader.read_many_sample(np.empty((1, 3)), number_of_samples, timeout=0) # Value not needed for demonstration
    except nidaqmx.DaqReadError as daq_read_error:
        # Check if task has already been stopped. If not, reraise error (else ignore).
        if not task.is_task_done():
            raise daq_read_error
        return -1 # return value of 1 indicates that error was ignored

    return 0

analog_reader = nidaqmx.stream_readers.AnalogMultiChannelReader(task.in_stream)
task.register_every_n_samples_acquired_into_buffer_event(1, _analog_samples_acquired_callback)

try:
    while True:
        print("START")
        task.start()
        print("STOP")
        task.stop()
finally:
    "DONE!"
    task.close()
```

The BSOD check code is either a 0x00000139 (KERNEL_SECURITY_CHECK_FAILURE) or 0x00000044 (MULTIPLE_IRP_COMPLETE_REQUESTS). Which one it is seems to be random (at least I don't know what causes the difference).
My original code was stopping and immediately restarting the task regularly. This is to ensure synchronization of the recording start time with an external signal, which happens every 4 to 9 seconds.
I'm attaching the contents of my 'C:\Windows\Minidump' folder as zip archive.

Any ideas on what the problem might be?

Thank you!

dnzrgn, 2024-09-16:
[dumps.zip](https://github.com/user-attachments/files/17011700/dumps.zip)

zhindes, 2024-09-16:
I took a look at those dumps. For anybody else looking
* 091624-17078-01.dmp - MULTIPLE_IRP_COMPLETE_REQUESTS - DAQ is not on the stack here, but that is expected.
* 091624-17312-01.dmp - KERNEL_SECURITY_CHECK_FAILURE - DAQ is on the stack here.

I haven't seen these types of failures in a VERY long time. Back in the early 2000s when there were a bunch of different USB Host Controller vendors, they were of... questionable(?) quality. I'm very surprised to be seeing them today.

So a question for you:
* What is your hardware setup? Specifically what is the USB Host Controller that this is plugged into. On Windows, you can check out Device Manager, find your DAQ card, then View >> Devices By Connection.
* Is this Windows in a VM?

A potential workaround (again from way back in the day...). Try setting [AIChannel.ai_usb_xfer_req_count](https://nidaqmx-python.readthedocs.io/en/stable/task_channels.html#nidaqmx.task.channels.AIChannel.ai_usb_xfer_req_count) to 1. This will limit streaming performance, but for the 6001, that almost certainly won't matter. You can also try setting [ai_usb_xfer_req_size](https://nidaqmx-python.readthedocs.io/en/stable/task_channels.html#nidaqmx.task.channels.AIChannel.ai_usb_xfer_req_size) to a smaller value than the default, 32768. Try powers of 2.

Regardless, this is a technical issue with your system and/or NI-DAQmx driver, not the DAQmx Python Bindings. That is a nuanced distinction, but it is important. I recommend posting on the [ni.com Multifunction DAQ Forum](https://forums.ni.com/t5/Multifunction-DAQ/bd-p/250). When you do, please post a link here. I'll follow along and chime in there.

closed 2024-09-16 by zhindes

dnzrgn, 2024-11-05:
Hello,

thank you for your advise! I posted the issue on the Multifunction DAQ Forum along with answers to your questions and outcome of your suggestions.

Here is the link: [https://forums.ni.com/t5/Multifunction-DAQ/Repeatedly-starting-and-stopping-analog-input-task-with/td-p/4406982](https://forums.ni.com/t5/Multifunction-DAQ/Repeatedly-starting-and-stopping-analog-input-task-with/td-p/4406982)