ni/nidaqmx-python #639: first_samp_timestamp_val property does not work because LibraryInterpreter is missing a method

bkeryan, 2024-10-03:
The `nidaqmx.Timing.first_samp_timestamp_val` property does not work because `LibraryInterpreter` is missing the `get_timing_attribute_timestamp` method.

```
>>> t = nidaqmx.Task()
>>> t.timing.first_samp_timestamp_val
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\dev\nidaqmx-python\generated\nidaqmx\task\_timing.py", line 424, in first_samp_timestamp_val
    val = self._interpreter.get_timing_attribute_timestamp(self._handle, 0x313a)
AttributeError: 'LibraryInterpreter' object has no attribute 'get_timing_attribute_timestamp'
```

bkeryan, 2024-10-03:
@zhindes @WayneDroid We should have added a test for this in https://github.com/ni/nidaqmx-python/pull/592 . Is there any reason we didn't?

Also, I think mypy would have caught this if the constructor arguments in https://github.com/ni/nidaqmx-python/blob/c09f2e286b8486272086d96f6a9846b20fa14520/generated/nidaqmx/task/_timing.py had type hints. We should try adding them, and if that enables mypy to catch errors like this, we should add these type hints to all of the subobject classes.

WayneDroid, 2024-10-04:
> @zhindes @WayneDroid We should have added a test for this in #592 . Is there any reason we didn't?
>
> Also, I think mypy would have caught this if the constructor arguments in [...]

I can't recall why no test case added for those properties, maybe overlooked.