ni/nidaqmx-python #640: Status Code -50405

keith331, 2024-10-08:
Hi Senior

I'm experiencing 50405 occurred when reading. The task only created once when app launch. and then reading io status when ever I want. It runs many times perfectly. but sometimes occurs that error. my nidaqmx version is 0.7.0.  I would like to update the version. but this equipment is in the other country. I'm worried about other side effects. Could somebody help Thanks in advance
` task_in = nidaqmx.Task()
  task_in.di_channels.add_di_chan('Dev1/port0/line0:7', line_grouping=LineGrouping.CHAN_PER_LINE)
  task_in.di_channels.add_di_chan('Dev1/port1/line0')`
  
 'call task_in.read()  frequently'

keith331, 2024-10-08:
Update:
NI said that it's related to USB power. I already used an independent USB port. Would It be helpful if I close task and re-create task regularly ?
https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z0000019MucSAE&l=zh-TW

zhindes, 2024-10-08:
Not nearly enough information to help you. Some questions:
* What version of NI-DAQmx driver is installed? Try running this example: [system_properties.py](https://github.com/ni/nidaqmx-python/blob/master/examples/system_properties.py)
* What device is "Dev1"?
* What does your actual code look like? Just paste it all.
* What Operating System are you using?