ni/nidaqmx-python #641: No devices found

bragostin, 2024-10-09:
That's an awesome package, thank you for developing it!  Unfortunately after many trials I still can't make it work:
```
Debian 12
python 3.11.2 (main, Aug 26 2024, 07:20:54) [GCC 12.2.0]
nidaqmx                  1.0.1
numpy                    1.24.2
```
`cat /run/nidevldu.status` yields:
```
2 Device(s) Found:

Device 1
Device Type: cDAQ-9174
Device Identifier: cDAQ1
Device GUID: 2d20ec26-861f-11ef-80ca-3ce1a1512ad8
Device Driver Type: MXDF
iDriver Class Name: nNICSR100::tMarlinDriver
iDriverInitializer Class Name: nNICSR100::tCDAQChassisDriverInitializer
iDriverClient Class Name: nNICSR100::tCDAQChassisDriverClient
iDriverFinalizer Class Name: nNICSR100::tCDAQChassisDriverFinalizer
tDriverBroker::createDriver status: 0
post createDriver status: 0
Device Init time: 599 ms
Device Batch Number: 2

Device 2
Device Type: NI 9264
Device Identifier: cDAQ1Mod4
Device GUID: 33a19a73-861f-11ef-80ca-3ce1a1512ad8
Device Driver Type: MXDF
iDriver Class Name: nNICSR100::tCDAQCartridgeDriver
iDriverInitializer Class Name: nNICSR100::tCDAQCartridgeDriverInitializer
iDriverClient Class Name: nNICSR100::tCDAQCartridgeDriverClient
iDriverFinalizer Class Name: 
tDriverBroker::createDriver status: 0
post createDriver status: 0
Device Init time: 561 ms
Device Batch Number: 3
```
It seems the device is there and recognized by the `nidaqmx` driver. 
However, `[dev.name for dev in nidaqmx.system.System.local().devices]` yields:
```
raceback (most recent call last):
  File "/home/chbrago/Documents/Labs/NI/Python/test.py", line 11, in <module>
    devices = [dev.name for dev in nidaqmx.system.System.local().devices]
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chbrago/Documents/Labs/NI/Python/test.py", line 11, in <listcomp>
    devices = [dev.name for dev in nidaqmx.system.System.local().devices]
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chbrago/cadquery/lib/python3.11/site-packages/nidaqmx/system/_collections/device_collection.py", line 71, in __iter__
    for device_name in self.device_names:
                       ^^^^^^^^^^^^^^^^^
  File "/home/chbrago/cadquery/lib/python3.11/site-packages/nidaqmx/system/_collections/device_collection.py", line 93, in device_names
    val = self._interpreter.get_system_info_attribute_string(0x193b)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chbrago/cadquery/lib/python3.11/site-packages/nidaqmx/_library_interpreter.py", line 3455, in get_system_info_attribute_string
    return value.value.decode(lib_importer.encoding)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: decode() argument 'encoding' must be str, not None
```
Any idea?

zhindes, 2024-10-09:
We get the locale [here](https://github.com/ni/nidaqmx-python/blob/c09f2e286b8486272086d96f6a9846b20fa14520/generated/nidaqmx/_lib.py#L212C9-L212C49). That uses [locale.getlocale()](https://docs.python.org/3.12/library/locale.html#locale.getlocale).

Can you open an interactive python prompt (just run python as is) and do this?
```
$ import locale
$ print(locale.getlocale())
```

bkeryan, 2024-10-09:
@zhindes This was also reported internally as AB#2827586. `locale.getlocale()` can return an `encoding` of `None`, so we ought to update nidaqmx-python to fall back to ASCII when the `encoding` is `None`.

zhindes, 2024-10-09:
@bkeryan - Ah nice, I'll try to find time this week to make a PR to address that!

@bragostin - Try setting your language environment variables as specified in the [locale man page](https://man7.org/linux/man-pages/man1/locale.1.html).

bragostin, 2024-10-10:
@zhindes `print(locale.getlocale())` yields `(None, None)`

closed 2024-10-14 by zhindes