ni/nidaqmx-python #644: nidaqmx doesn't support Python 3.13+

zhindes, 2024-10-16:
To reproduce:
* Install Python 3.13
* `pip install nidaqmx`
* Run some of the shipping examples

At the very least I ran into this:
```
Traceback (most recent call last):
  File "C:\Users\User\dev\junyinta-nimbus-benchmark\ramp_up.py", line 5, in <module>
    import nidaqmx
  File "C:\Users\User\AppData\Local\pypoetry\Cache\virtualenvs\benchmark-PLHonyti-py3.13\Lib\site-packages\nidaqmx\__init__.py", line 3, in <module>
    from nidaqmx.scale import Scale
  File "C:\Users\User\AppData\Local\pypoetry\Cache\virtualenvs\benchmark-PLHonyti-py3.13\Lib\site-packages\nidaqmx\scale.py", line 3, in <module>
    import numpy
ModuleNotFoundError: No module named 'numpy'
```

That is because our numpy dependency excludes Python 3.13+ [here](https://github.com/ni/nidaqmx-python/blob/41cb6ba35c0781dec34d2e4fc91e23b22b813135/pyproject.toml#L35). There may be other issues, though. We should officially support and test Python 3.13+

bkeryan, 2024-10-16:
Oops, I should have made that an open-ended range or capped it at 4.0 like I did for grpcio-tools: https://github.com/ni/nidaqmx-python/pull/453/

bkeryan, 2024-10-16:
Note that we need to add Python 3.13 to the nidaqmxbot image build config.

PR #645 (2024-10-16): nidaqmx: Fix installation with Python 3.13
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nidaqmx-python/blob/master/CONTRIBUTING.md).

### What does this Pull Request accomplish?

Update the dependency versions in `pyproject.toml` to be compatible with Python 3.13. For per-Python version constraints, make sure the last version is `^3.13` so that it will also cover 3.14, etc.

Loosen `protobuf` version constraint to allow `protobuf` 5.x because `grpcio-tools` 1.67 requires `protobuf` 5.x.

Enable running unit tests with Python 3.13 for the PR/CI workflows. I didn't  [... cut]
PR #650 (2024-10-17): [releases/1.0] Add support for Python 3.13
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nidaqmx-python/blob/master/CONTRIBUTING.md).

### What does this Pull Request accomplish?

Cherry-pick support for Python 3.13:
- #645 
- #646 
- #649 

### Why should this Pull Request be merged?

Fixes #644 

### What testing has been done?

See linked PRs.
closed 2024-10-16 by bkeryan