ni/nidaqmx-python #651: Information on callback execution

gabcavic, 2024-10-29:
Hi,
I am building an application including real-time *streaming* of data from the DAQ. I would need some information on the execution models of callbacks such as those provided to functions `register_every_n_samples_acquired_into_buffer_event`. What happens when the task is running and this callback is executed? Is there a parallel thread launching the event and executing the callback? Is it a blocking execution?

I couldn't find anything in the documentation, but since the execution model in Python can be various (multiprocessing, asynchronous, thread-based, etc) it would be nice to have some clarification on this point.

zhindes, 2024-10-29:
The implementation is this:
* [task.py::register_every_n_samples_acquired_into_buffer_event](https://github.com/ni/nidaqmx-python/blob/ba8a96232f67384c345904290040fe3a448dc5e8/src/handwritten/task/_task.py#L809)
* [_library_interpreter.py::_library_interpreter.py](https://github.com/ni/nidaqmx-python/blob/ba8a96232f67384c345904290040fe3a448dc5e8/generated/nidaqmx/_library_interpreter.py#L4667)
* **NIDAQmx C API, DAQmxRegisterEveryNSamplesEvent** - documentation on ni.com [here](https://www.ni.com/docs/en-US/bundle/ni-daqmx-c-api-ref/page/daqmxcfunc/daqmxregistereverynsamplesevent.html?srsltid=AfmBOoqfzIfCmWllqcP5NK4D_C7cR5vPLW3rZfwM-HPQTL2aISlDCojB)

You can see that we pass `0` for the options parameter in task.py. The documentation says

> The callback function is called in a DAQmx thread. This is the default value.

If this is useful, feel free to make a PR to `task.py` to update the documentation! If there are more questions you have, let me know. I can dig into the internal code deeper.

gabcavic, 2024-11-01:
THank you very much for the answer! As soon as I will have a bit of time I will prepare the PR