ni/nidaqmx-python #666: Unable to debug the nidaqmx related code in disk D

Nocturne-7, 2024-12-30:
Hi,
I downloaded the example code to my computer desktop and had no problem running or debugging it. However, when I saved the code file to the path D: \, the code could run directly but could not be debugged. Excuse me, where is the problem? Is there something unconfigured?
I am using the Python 3.8 environment created by Anaconda

[image]

zhindes, 2025-01-02:
Your problem is almost certainly unrelated to the nidaqmx-python module, and it is, instead, related to some sort of configuration issue with your Anaconda environments or Visual Studio. Perhaps [Select a Python environment for a project in Visual Studio](https://learn.microsoft.com/en-us/visualstudio/python/selecting-a-python-environment-for-a-project?view=vs-2022) might be relevant? I'd recommend Stack Overflow if you can't resolve it based on that.

Good luck!

closed 2025-01-02 by zhindes

bkeryan, 2025-01-06:
The stack trace in the screenshot indicates that the problem is related to the [python-decouple](https://pypi.org/project/python-decouple/) module, which nidaqmx-python uses to read feature toggles from environment variables or a `.env` file.

This python-decouple bug describes a similar issue with IPython, where `_find_file` recurses infinitely when running a script located on a different drive letter: https://github.com/HBNetwork/python-decouple/issues/172