ni/nidaqmx-python #692: Cannot set "ai_conv_rate" on NI DAQ 9209

CleSpi, 2025-03-27:
To avoid ghosting during switching, we need to increase the ai_conv_rate. However, the standard command "task.timing.ai_conv_rate = 1.1 * sampling_rate * n_channels" does not work on the DAQ model 9209.

zhindes, 2025-03-27:
This is likely because setting the convert rate on a cDAQ module requires specifying an active device on the timing property, which the Python API does not support.

Quoting some internal information:

As a workaround, it is possible to use the internal "interpreter" interface to call the `set_timing_attribute_ex_double method`, but this is not a stable API and future versions of nidaqmx-python may break compatibility with it: https://github.com/search?q=repo%3Ani/nidaqmx-python%20set_timing_attribute_ex_double&type=code

```python
# Unsupported workaround, may break in future releases
task._interpreter.set_timing_attribute_ex_double(task._handle, "cDAQ1Mod1", 0x1848, 100.0)
```

CleSpi, 2025-03-27:
Yes, I am aware of this workaround. However, as you mention, it may not work in future versions.

bkeryan, 2025-03-27:
If we follow the design of the DAQmx .NET API, implementing this feature will involve making the [Timing](https://github.com/ni/nidaqmx-python/blob/master/generated/nidaqmx/task/_timing.py) class indexable with a string:

```python
task.timing["cDAQ1Mod1"].ai_conv_rate = 1.1 * sampling_rate * n_channels
```

We would add an `_active_devs` field to the `Timing` class and a `Timing.__getitem__` method that returns a new instance of `Timing` with `_active_devs` set to the index. All of the get/set/reset methods in the `Timing` class would have a condition to call the `get|set_reset_timing_attribute_ex` method if `_active_devs` is set. We wouldn't want to use the `ex` methods in the normal case because they support only a very small number of attributes (basically the AI convert clock and delay from sample clock attributes).

davitdan, 2025-03-27:
@bkeryan And if I am not mistaken, the getter for this func is not implemented either right? (get_timing_attribute_ex_double)

bkeryan, 2025-03-27:
@davitdan It's implemented for `LibraryInterpreter`: https://github.com/ni/nidaqmx-python/blob/master/generated/nidaqmx/_library_interpreter.py#L3572
and `GrpcInterpreter`: https://github.com/ni/nidaqmx-python/blob/master/generated/nidaqmx/_grpc_interpreter.py#L2135

`BaseInterpreter` is an abstract base class, so all of its methods are stubs that raise `NotImplementedError`.

PR #767 (2025-07-15): Add Active Dev to Timing Class
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nidaqmx-python/blob/master/CONTRIBUTING.md).
- [x] I've updated [CHANGELOG.md](https://github.com/ni/nidaqmx-python/blob/master/CHANGELOG.md) if applicable.
- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?
- Update attributes.py metadata generated from [add lv-filters to attribute metadata by charitylxy · Pull Request #269 · ni/grpc-device-scrapigen](https://github.com/ni/grpc-device-scrapigen/pull/269)
- Add `_active_devs` to `Timing` cl [... cut]
closed 2025-07-15 by charitylxy